gaoxiang12/slambook #187: 感觉公式(10.33)好像不对。

jayoungo, 2019-03-08:
高博，您好！
第243页公式(10.33)，类似公式(10.31)的推导，感觉应该是：
`... = N(h(x^-_k), HP^^_kH^T + Q_k)`
或者是我理解不对？
谢谢！

jayoungo, 2019-03-11:
我理解的是，由观测方程(10.29)：
[image]
右边第2项在求均值的时候等于0（括号内第1项的均值等于第2项），求协方差时括号内的第2项由于是常数所以得0，第1项得：
[image]
所以式(10.33)应该是：
[image]
是这样吗？谁能帮我解答一下。谢谢！

jayoungo, 2019-03-14:
@gaoxiang12 高博，您好！能帮我看一下这个问题吗？

gaoxiang12, 2019-03-14:
Sorry啊我现在手边的书送人了……等我重新买一本再来回复你。

jayoungo, 2019-03-14:
奥，好的。谢谢！

gaoxiang12, 2019-04-10:
@jayoungo 我看了一下，这里 P(z_k|x_k) 是固定 x_k 的，所以除了n_k之外其他的都是固定量，这个式子应该还是按照书上的写法。但这步也只是个中间结果，不影响结论。
感谢指出！

jayoungo, 2019-04-10:
@gaoxiang12 明白了。谢谢高博！

closed 2019-04-10 by jayoungo