gaoxiang12/slambook #188: 关于P196，8.16直接法误差对李代数的导数推导

qq456cvb, 2019-03-09:
这部分虽然结果一样，但我感觉推导过程存在问题，非常令人困惑。我的推导如下：
令
<a href="https://www.codecogs.com/eqnedit.php?latex=u&space;=&space;\frac{1}{Z_2}K\mathrm{exp}(\xi^\wedge)P" target="_blank"><img src="https://latex.codecogs.com/gif.latex?u&space;=&space;\frac{1}{Z_2}K\mathrm{exp}(\xi^\wedge)P" title="u = \frac{1}{Z_2}K\mathrm{exp}(\xi^\wedge)P" /></a>，
<a href="https://www.codecogs.com/eqnedit.php?latex=q=\mathrm{exp}(\xi^\wedge)P" target="_blank"><img src="https://latex.codecogs.com/gif.latex?q=\mathrm{exp}(\xi^\wedge)P" title="q=\mathrm{exp}(\xi^\wedge)P" /></a>，
那么
<a href="https://www.codecogs.com/eqnedit.php?latex=\begin{align*}&space;e(\xi&plus;\delta\xi)&=I_1&space;-&space;I_2(\xi&plus;\delta\xi)\\&space;&=I_1-[I_2(\xi)&plus;\frac{\partial&space;I_2}{\partial\xi}\cdot\delta\xi]\\&space;&=I_1-I_2(\xi)-\frac{\partial&space;I_2}{\partial&space;u}\frac{\partial&space;u}{\partial&space;q}\frac{\partial&space;q}{\partial\xi}\cdot\delta\xi\\&space;&=e(\xi)-\frac{\partial&space;I_2}{\partial&space;u}\frac{\partial&space;u}{\partial&space;q}\frac{\partial&space;q}{\partial\xi}\cdot\delta\xi&space;\end{align*}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\begin{align*}&space;e(\xi&plus;\delta\xi)&=I_1&space;-&space;I_2(\xi&plus;\delta\xi)\\&space;&=I_1-[I_2(\xi)&plus;\frac{\partial&space;I_2}{\partial\xi}\cdot\delta\xi]\\&space;&=I_1-I_2(\xi)-\frac{\partial&space;I_2}{\partial&space;u}\frac{\partial&space;u}{\partial&space;q}\frac{\partial&space;q}{\partial\xi}\cdot\delta\xi\\&space;&=e(\xi)-\frac{\partial&space;I_2}{\partial&space;u}\frac{\partial&space;u}{\partial&space;q}\frac{\partial&space;q}{\partial\xi}\cdot\delta\xi&space;\end{align*}" title="\begin{align*} e(\xi+\delta\xi)&=I_1 - I_2(\xi+\delta\xi)\\ &=I_1-[I_2(\xi)+\frac{\partial I_2}{\partial\xi}\cdot\delta\xi]\\ &=I_1-I_2(\xi)-\frac{\partial I_2}{\partial u}\frac{\partial u}{\partial q}\frac{\partial q}{\partial\xi}\cdot\delta\xi\\ &=e(\xi)-\frac{\partial I_2}{\partial u}\frac{\partial u}{\partial q}\frac{\partial q}{\partial\xi}\cdot\delta\xi \end{align*}" /></a>。
这里将<a href="https://www.codecogs.com/eqnedit.php?latex=I_2" target="_blank"><img src="https://latex.codecogs.com/gif.latex?I_2" title="I_2" /></a>看成是<a href="https://www.codecogs.com/eqnedit.php?latex=\xi" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\xi" title="\xi" /></a>的导数，然后用一阶泰勒展开。而文中的一阶泰勒展开让人很困惑。

另外，我建议，扰动模型的导数在表示时在分母上用<a href="https://www.codecogs.com/eqnedit.php?latex=\partial\xi" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\partial\xi" title="\partial\xi" /></a>而不是<a href="https://www.codecogs.com/eqnedit.php?latex=\partial\delta\xi" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\partial\delta\xi" title="\partial\delta\xi" /></a>，否则很让人很困惑，感觉像是在对增量求导，展开后却又不是。

gaoxiang12, 2019-03-11:
了解。这部分其实和上一章一样，只是多了个图像梯度。

MaxTianyy, 2019-04-26:
good, 今天也发现这个问题，书上的推导确实很困惑，这样写比较清晰。

lichenyue1996, 2019-08-22:
你好，我还想问下这部分的direct_sparse代码中求的jacobian_uv_ksai是否有问题，看起来跟推导中的公式里456列和123列反了，或者是有什么原因吗？

lanfeiying, 2020-12-06:
@gaoxiang12 
你好，这里泰勒展开我也看不太明白，这个部分在最新的书上修订了吗？
这个泰勒展开应该怎么理解呢，是对 \xi 还是 \delta\xi 求导呢？