gaoxiang12/slambook #190: ch4编译Sophus时编译不通过

GuoBoyang, 2019-03-19:
By not providing "FindSophus.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Sophus", but
  CMake did not find one.

  Could not find a package configuration file provided by "Sophus" with any
  of the following names:

    SophusConfig.cmake
    sophus-config.cmake

  Add the installation prefix of "Sophus" to CMAKE_PREFIX_PATH or set
  "Sophus_DIR" to a directory containing one of the above files.  If "Sophus"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/slambook/ch4/CMakeFiles/CMakeOutput.log".

GuoBoyang, 2019-03-19:
How to solve this problem ?

AnfangRobkit, 2019-08-17:
You should first build the Sophus:
cd Sophus
cmake .
make