gaoxiang12/slambook #193: ch11中编译pose_grapg_gtsam出错,麻烦哪位大神帮我看一下这是什么问题啊

Treffen, 2019-04-11:
jys236@236:~/slambook/ch11/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CHOLMOD: /usr/include/suitesparse  
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:45 (message):
  Invalid arguments to find_dependency
Call Stack (most recent call first):
  /usr/local/lib/cmake/GTSAM/GTSAMConfig.cmake:17 (find_dependency)
  CMakeLists.txt:25 (find_package)

Zkjoker, 2019-05-03:
我同样有此问题 无法解决，系统是ubuntu16.04的 请问您解决了吗

Treffen, 2019-05-04:
解决了，我将错误提示的find_dependency改为了find_package就好了

Zkjoker, 2019-05-04:
谢谢！！