gaoxiang12/slambook #197: CH12 make error

ZENGjunxian, 2019-04-19:
make[2]: *** No rule to make target '/usr/local/lib/libDBoW3.a', needed by 'gen_vocab'。 停止。
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/gen_vocab.dir/all' failed
make[1]: *** [CMakeFiles/gen_vocab.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
I don't know how to use this ch12.I beg you may say more in detail

IvyGongoogle, 2019-08-11:
meet the same error. Have you solved it ? @ZENGjunxian

IvyGongoogle, 2019-08-12:
specify the correct `DBoW3_LIBS` in CMakeLists.txt

amituofo1996, 2020-09-08:
please modify  feature_trainning.cpp ,if it exist libDBoW3.a',please change it with libDBoW3.so.

salihmarangoz, 2021-04-14:
Change this line in ch11/CMakeLists.txt:
**set( DBoW3_LIBS "/usr/local/lib/libDBoW3.a" )**
to:
**set( DBoW3_LIBS "/usr/local/lib/libDBoW3.so" )**

iamkrs9, 2021-05-22:
I replaced it with /usr/local/lib/libDBoW3.dylib and it worked. I am using MacOS 11.3