gaoxiang12/slambook #208: ch10 g2o_custombundle 编译报错

suzhengpeng, 2019-05-15:
[image]
如上图所示。g2o的版本是 https://github.com/RainerKuemmerle/g2o 下载的最新版。

gaoxiang12, 2019-05-16:
请使用本仓库自带的g2o版本。本书出版之后g2o有较大更新，主要将祼指针接口改成了智能指针，所以原先代码在新版本中编译不过。

chauchy-Liu, 2022-11-25:
Undefined symbols for architecture x86_64:
[build]   "g2o::HyperGraph::Edge::resize(unsigned long)", referenced from:
[build]       g2o::BaseFixedSizedEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1>, VertexCameraBAL, VertexPointBAL>::resize(unsigned long) in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1> > in g2o_bundle.cpp.o
[build]   "g2o::HyperGraph::vertex(int)", referenced from:
[build]       g2o::OptimizableGraph::vertex(int) in g2o_bundle.cpp.o
[build]   "g2o::RobustKernel::RobustKernel()", referenced from:
[build]       g2o::RobustKernelHuber::RobustKernelHuber() in g2o_bundle.cpp.o
[build]   "g2o::writeCCSMatrix(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int, int, int const*, int const*, double const*, bool)", referenced from:
[build]       g2o::LinearSolverCholmod<Eigen::Matrix<double, 9, 9, 0, 9, 9> >::saveMatrix(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in g2o_bundle.cpp.o
[build]   "g2o::MatrixStructure::alloc(int, int)", referenced from:
[build]       g2o::SparseBlockMatrix<Eigen::Matrix<double, 9, 9, 0, 9, 9> >::fillBlockStructure(g2o::MatrixStructure&) const in g2o_bundle.cpp.o
[build]   "g2o::MatrixStructure::MatrixStructure()", referenced from:
[build]       g2o::LinearSolverCholmod<Eigen::Matrix<double, 9, 9, 0, 9, 9> >::LinearSolverCholmod() in g2o_bundle.cpp.o
[build]   "g2o::MatrixStructure::~MatrixStructure()", referenced from:
[build]       g2o::LinearSolverCholmod<Eigen::Matrix<double, 9, 9, 0, 9, 9> >::LinearSolverCholmod() in g2o_bundle.cpp.o
[build]       g2o::LinearSolverCholmod<Eigen::Matrix<double, 9, 9, 0, 9, 9> >::~LinearSolverCholmod() in g2o_bundle.cpp.o
[build]   "g2o::SparseOptimizer::setVerbose(bool)", referenced from:
[build]       SolveProblem(char const*, BundleParams const&) in g2o_bundle.cpp.o
[build]   "g2o::SparseOptimizer::setAlgorithm(g2o::OptimizationAlgorithm*)", referenced from:
[build]       SetSolverOptionsFromFlags(BALProblem*, BundleParams const&, g2o::SparseOptimizer*) in g2o_bundle.cpp.o
[build]   "g2o::SparseOptimizer::initializeOptimization(int)", referenced from:
[build]       SolveProblem(char const*, BundleParams const&) in g2o_bundle.cpp.o
[build]   "g2o::SparseOptimizer::optimize(int, bool)", referenced from:
[build]       SolveProblem(char const*, BundleParams const&) in g2o_bundle.cpp.o
[build]   "g2o::SparseOptimizer::SparseOptimizer()", referenced from:
[build]       SolveProblem(char const*, BundleParams const&) in g2o_bundle.cpp.o
[build]   "g2o::SparseOptimizer::~SparseOptimizer()", referenced from:
[build]       SolveProblem(char const*, BundleParams const&) in g2o_bundle.cpp.o
[build]   "g2o::OptimizableGraph::Edge::resolveCaches()", referenced from:
[build]       vtable for EdgeObservationBAL in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseBinaryEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1>, VertexCameraBAL, VertexPointBAL> in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseFixedSizedEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1>, VertexCameraBAL, VertexPointBAL> in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1> > in g2o_bundle.cpp.o
[build]   "g2o::OptimizableGraph::Edge::setRobustKernel(g2o::RobustKernel*)", referenced from:
[build]       BuildProblem(BALProblem const*, g2o::SparseOptimizer*, BundleParams const&) in g2o_bundle.cpp.o
[build]   "g2o::OptimizableGraph::Edge::setMeasurementData(double const*)", referenced from:
[build]       vtable for EdgeObservationBAL in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseBinaryEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1>, VertexCameraBAL, VertexPointBAL> in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseFixedSizedEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1>, VertexCameraBAL, VertexPointBAL> in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1> > in g2o_bundle.cpp.o
[build]   "g2o::OptimizableGraph::Edge::setMeasurementFromState()", referenced from:
[build]       vtable for EdgeObservationBAL in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseBinaryEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1>, VertexCameraBAL, VertexPointBAL> in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseFixedSizedEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1>, VertexCameraBAL, VertexPointBAL> in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1> > in g2o_bundle.cpp.o
[build]   "g2o::OptimizableGraph::Edge::Edge()", referenced from:
[build]       g2o::BaseEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1> >::BaseEdge() in g2o_bundle.cpp.o
[build]   "g2o::OptimizableGraph::Edge::~Edge()", referenced from:
[build]       g2o::BaseEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1> >::BaseEdge() in g2o_bundle.cpp.o
[build]       g2o::BaseEdge<2, Eigen::Matrix<double, 2, 1, 0, 2, 1> >::~BaseEdge() in g2o_bundle.cpp.o
[build]   "g2o::OptimizableGraph::Vertex::updateCache()", referenced from:
[build]       vtable for VertexCameraBAL in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseVertex<9, Eigen::Matrix<double, -1, 1, 0, -1, 1> > in g2o_bundle.cpp.o
[build]       vtable for VertexPointBAL in g2o_bundle.cpp.o
[build]       vtable for g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> > in g2o_bundle.cpp.o
[build]   "g2o::OptimizableGraph::Vertex::Vertex()", referenced from:
[... cut ...]
[build] ld: symbol(s) not found for architecture x86_64
[build] clang: error: linker command failed with exit code 1 (use -v to see invocation)
[build] make[2]: *** [/Users/liuchuanxi/slam/ch10/g2o_custombundle/bin/g2o_customBundle] Error 1
[build] make[1]: *** [CMakeFiles/g2o_customBundle.dir/all] Error 2
[build] make: *** [all] Error 2
[proc] The command: /usr/local/bin/cmake --build /Users/liuchuanxi/slam/ch10/build --config Debug --target all -- exited with code: 2 and signal: null
[build] Build finished with exit code 2