gaoxiang12/slambook #216: ch5 joinMap两个编译错误和解决办法

zergzzlun, 2019-06-07:
- Ch5 joinMap部分make时遇到两个报错：
`/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libSM.so: undefined reference to uuid_unparse_lower@UUID_1.0'`
这个报错可以通过在`CMAKE_CXX_FLAGS`里加入 `-luuid` 解决。

- 另一个报错是：
`ImportError: libfontconfig.so.1: undefined symbol: FT_Done_MM_Var`
根据[这里](https://github.com/opencv/opencv/issues/12625)里的经验删除一部分Anaconda里的文件即可解决这个问题，即：
```
<...>/anaconda3/lib/libfontconfig.so
<...>/anaconda3/lib/libfontconfig.so.1
<...>/anaconda3/lib/libfontconfig.so.1.11.1
<...>/anaconda3/lib/libpangoft2-1.0.so.0
<...>/anaconda3/lib/libpangoft2-1.0.so.0.4200.4
```

MavAlex, 2019-09-25:
[ 50%] Building CXX object CMakeFiles/joinMap.dir/joinMap.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[100%] Linking CXX executable joinMap
/usr/bin/ld: cannot find -lvtkproj4
collect2: error: ld returned 1 exit status
CMakeFiles/joinMap.dir/build.make:349: recipe for target 'joinMap' failed
make[2]: *** [joinMap] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/joinMap.dir/all' failed
make[1]: *** [CMakeFiles/joinMap.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
*** Failure: Exit code 2 ***
老哥make的时候出现这个问题怎么回事？

zergzzlun, 2019-09-26:
Hi Alex,

我并不完全了解你遇到的这个问题，
根据这个和你遇到了类似报错的issue:
https://github.com/PointCloudLibrary/pcl/issues/1594
一个可能的解决办法是：
`ln -s /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.2.so /usr/lib/libvtkproj4.so`

希望可以帮助你。