gaoxiang12/slambook #231: ch7 关于g2o优化输出信息verbose中 具体信息代表什么意思？如chi2，edges， schur， lambda

valian1982, 2019-08-09:
高博 您好。
g2o优化信息输出通过 optimizer.setVerbose( true ); 输出。
根据输出信息显示包括以下部分：iteration= 0	 chi2= 18115.373754	 time= 8.537e-05	 cumTime= 8.537e-05	 edges= 74	 schur= 0	 lambda= 7.952906	 levenbergIter= 1

高博，这些信息在哪里可以查到他们具体代表的什么意思？我查了g2o的很多c源码，都没有找到。谢谢您：）

LongruiDong, 2020-07-17:
在/g2o/g2o/core/sparse_optimizer.cpp里有相关代码
```
      if (verbose()){
        number_t dts = get_monotonic_time()-ts;
        cumTime += dts;
        if (! errorComputed)
          computeActiveErrors();
        //verbose print  out is here!!
        cerr << "iteration= " << i
          << "\t chi2= " << FIXED(activeRobustChi2())
          << "\t time= " << dts
          << "\t cumTime= " << cumTime
          << "\t edges= " << _activeEdges.size();
        _algorithm->printVerbose(cerr);
        cerr << endl;
      }
```

LongruiDong, 2020-07-17:
> 高博 您好。
> g2o优化信息输出通过 optimizer.setVerbose( true ); 输出。
> 根据输出信息显示包括以下部分：iteration= 0 chi2= 18115.373754 time= 8.537e-05 cumTime= 8.537e-05 edges= 74 schur= 0 lambda= 7.952906 levenbergIter= 1
> 
> 高博，这些信息在哪里可以查到他们具体代表的什么意思？我查了g2o的很多c源码，都没有找到。谢谢您：）

我在/g2o/g2o/core/sparse_optimizer.cpp里看到有相关代码
```
      if (verbose()){
        number_t dts = get_monotonic_time()-ts;
        cumTime += dts;
        if (! errorComputed)
          computeActiveErrors();
        //verbose print  out is here!!
        cerr << "iteration= " << i
          << "\t chi2= " << FIXED(activeRobustChi2())
          << "\t time= " << dts
          << "\t cumTime= " << cumTime
          << "\t edges= " << _activeEdges.size();
        _algorithm->printVerbose(cerr);
        cerr << endl;
      }
```