gaoxiang12/slambook #233: project/0.4,visual_odometry.cpp中关于地图上的特征点是否在当前帧的判断问题

xilongWang, 2019-08-19:
@gaoxiang12 高博您好，在visual_odometry.cpp文件的featureMatching()函数中，会对地图上的特征点进行判断其是否出现在当前帧中，在isInFrame()函数中，使用的是参考坐标系的T_c_w_，而在地图(特征点集合)中，并不是每一个特征点都来自相同的某一帧，这样计算出来的isInFrame()感觉不太准确，我认为每一个特征点都应该有一个T_c_w_，这样isInFrame()计算才较为准确，不知道我的理解是否正确？

xilongWang, 2019-08-20:
我按照我的理解对isInFrame()进行更改后，可以跑完整个数据集！

dcbin, 2025-03-04:
> [@gaoxiang12](https://github.com/gaoxiang12) 高博您好，在visual_odometry.cpp文件的featureMatching()函数中，会对地图上的特征点进行判断其是否出现在当前帧中，在isInFrame()函数中，使用的是参考坐标系的T_c_w_，而在地图(特征点集合)中，并不是每一个特征点都来自相同的某一帧，这样计算出来的isInFrame()感觉不太准确，我认为每一个特征点都应该有一个T_c_w_，这样isInFrame()计算才较为准确，不知道我的理解是否正确？

不是的，这里是用上一帧的相机位姿粗略判断局部地图中的点是否在当前帧中，地图中的特征点都是在世界坐标系下，乘个T_c_w_就能把每一个特征点转换到相机坐标系下，而不是说每一个特征点都有一个T_c_w_，这里的T_c_w是相机与世界坐标的相对位姿，与地图中的特征点没有关系，改了反而不对。