gaoxiang12/slambook #239: ch5 joinmap 编译问题

MavAlex, 2019-09-25:
同kedevelop build之后显示：
[ 50%] Building CXX object CMakeFiles/joinMap.dir/joinMap.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[100%] Linking CXX executable joinMap
/usr/bin/ld: cannot find -lvtkproj4
collect2: error: ld returned 1 exit status
CMakeFiles/joinMap.dir/build.make:349: recipe for target 'joinMap' failed
make[2]: *** [joinMap] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/joinMap.dir/all' failed
make[1]: *** [CMakeFiles/joinMap.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
*** Failure: Exit code 2 ***
 求解，卡在这里好久了

MavAlex, 2019-09-25:
/usr/local/include/pcl-1.8/pcl/visualization/point_cloud_geometry_handlers.h:48:29: fatal error: vtkSmartPointer.h: No such file or directory
compilation terminated.
CMakeFiles/joinMap.dir/build.make:62: recipe for target 'CMakeFiles/joinMap.dir/joinMap.cpp.o' failed
make[3]: *** [CMakeFiles/joinMap.dir/joinMap.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/joinMap.dir/all' failed
make[2]: *** [CMakeFiles/joinMap.dir/all] Error 2
CMakeFiles/Makefile2:79: recipe for target 'CMakeFiles/joinMap.dir/rule' failed
make[1]: *** [CMakeFiles/joinMap.dir/rule] Error 2
Makefile:118: recipe for target 'joinMap' failed
make: *** [joinMap] Error 2
*** Failure: Exit code 2 ***
加上高博那一句话就变成了这样，求解 。。。

tiemuhua, 2020-08-24:
你看看你能不能找到libvtkproj4.so这个文件，可能在/usr里面也有可能在编译出来的东西里面
用find / -name libvtkproj4.so命令找。
能找到的话复制一份放到/usr/lib里面。
再编译