gaoxiang12/slambook #240: 第九章0.2在执行到

jiujiuone, 2019-10-06:
在程序执行到string dataset_dir = myslam::Config::get<string> ( "dataset_dir" );时出现Segmentation fault (core dumped)错误
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff6bfc94a in cv::FileNode::ptr() [clone .localalias.329] ()
   from /usr/local/lib/libopencv_core.so.4.1
(gdb) where
#0  0x00007ffff6bfc94a in cv::FileNode::ptr() [clone .localalias.329] ()
   from /usr/local/lib/libopencv_core.so.4.1
#1  0x00007ffff6bfca39 in cv::FileNode::type() const ()
   from /usr/local/lib/libopencv_core.so.4.1
#2  0x00007ffff6bfca99 in cv::FileNode::isMap() const ()
   from /usr/local/lib/libopencv_core.so.4.1
#3  0x00007ffff6c01db1 in cv::FileNode::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const ()
   from /usr/local/lib/libopencv_core.so.4.1
#4  0x00007ffff6c01f61 in cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const ()
   from /usr/local/lib/libopencv_core.so.4.1
#5  0x0000555555554fa9 in main ()
怀疑和config.h里的模板get函数有关系，却不知道怎么解。请大神指教。

fengweigang, 2020-06-08:
> 在程序执行到string dataset_dir = myslam::Config::get ( "dataset_dir" );时出现Segmentation fault (core dumped)错误
> Program received signal SIGSEGV, Segmentation fault.
> 0x00007ffff6bfc94a in cv::FileNode::ptr() [clone .localalias.329] ()
> from /usr/local/lib/libopencv_core.so.4.1
> (gdb) where
> #0 0x00007ffff6bfc94a in cv::FileNode::ptr() [clone .localalias.329] ()
> from /usr/local/lib/libopencv_core.so.4.1
> #1 0x00007ffff6bfca39 in cv::FileNode::type() const ()
> from /usr/local/lib/libopencv_core.so.4.1
> #2 0x00007ffff6bfca99 in cv::FileNode::isMap() const ()
> from /usr/local/lib/libopencv_core.so.4.1
> #3 0x00007ffff6c01db1 in cv::FileNode::operator[](std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&) const ()
> from /usr/local/lib/libopencv_core.so.4.1
> #4 0x00007ffff6c01f61 in cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&) const ()
> from /usr/local/lib/libopencv_core.so.4.1
> #5 0x0000555555554fa9 in main ()
> 怀疑和config.h里的模板get函数有关系，却不知道怎么解。请大神指教。

I meet the problem like yours. it seems like yaml file read failed. but I can‘t figure out why. Have you solved the problem? I runn in windows environment.

mitsurukato24, 2020-08-28:
I also met the error from myslam::Config::get().
I'm using Visual Studio 2015 and OpenCV 4.3.

I changed the code of the function Config::setParameterFile(const std::string& filename) in config.cpp.

```
// config_->file_ = cv::FileStorage( filename.c_str(), cv::FileStorage::READ );
config_->file_.open(filename.c_str(), cv::FileStorage::READ);
```

After I changed this, Config::get() worked properly in my environment so can you try this?

YinpeiDai, 2023-11-20:
Thank you so much @mitsurukato24 , it works