gaoxiang12/slambook #246: OpenCV Error: Assertion failed (TlsSetValue(tlsKey, pData) == TRUE)

XGuider, 2019-12-17:
环境：win10+opencv3.1.0+vs2013
运行问题：运行本例ch7 中的feature_extraction.cpp，发现程序会在[    Ptr<FeatureDetector> detector = ORB::create();
    Ptr<DescriptorExtractor> descriptor = ORB::create();
    // Ptr<FeatureDetector> detector = FeatureDetector::create(detector_name);
    // Ptr<DescriptorExtractor> descriptor = DescriptorExtractor::create(descriptor_name);
    Ptr<DescriptorMatcher> matcher  = DescriptorMatcher::create ( "BruteForce-Hamming" );

    //-- 第一步:检测 Oriented FAST 角点位置
    detector->detect ( img_1,keypoints_1 );   -------->>>>这里报错，错误类型为上面的提示
    detector->detect ( img_2,keypoints_2 );
](url)
谢谢作者的工作，困扰很久，希望解答！！！！谢谢

gaoxiang12, 2019-12-18:
你的图片貌似没传上来，不过我推测是img_1没读进来的原因。请检查下路径是否正确。

XGuider, 2019-12-18:
谢谢博主的及时的回答，
`	Mat img_1 = imread("1.png", CV_LOAD_IMAGE_COLOR);
	Mat img_2 = imread("2.png", CV_LOAD_IMAGE_COLOR);
	//Mat img_1 = imread(argv[0], CV_LOAD_IMAGE_COLOR);
	//Mat img_2 = imread(argv[1], CV_LOAD_IMAGE_COLOR);
	//std::cout << img_1 << std::endl;------------------->输出图像矩阵
	//std::cout << img_2 << std::endl;`

但是，我这输出img_1是有数据输出的
我怀疑是不是我环境的问题，导致这个问题？但是不清楚哪出现问题，（没有安装opencv contrib）