gaoxiang12/slambook #251: ch12 回环检测 使用vocab_larger字典结果不准确

Stark-Chang, 2020-04-10:
loop_closure.cpp DBoW3::Vocabulary vocab("./vocab_larger.yml.gz");使用的是文件夹中自带的vocab_larger字典，没有用gen_vocab_large.cpp。结果很不准确，以第十张图为例：
searching for image 9 returns 4 results:
<EntryId: 9, Score: 1>
<EntryId: 6, Score: 0.0445693>
<EntryId: 3, Score: 0.042178>
<EntryId: 5, Score: 0.0419668>
明显跟第一张图没有相似性啊，请问是怎么回事啊。

Ryyyc, 2023-05-07:
我猜测你使用的OpenCV版本和 本仓库生成vocab_larger.yml.gz时使用的不一致，也就是说，对于同一张图像，你生成的ORB特征和本仓库的不同。这种情况直接去跑相似度计算得不到想要的结果。

BTREE-C802, 2023-05-07:
您好，我叫旷鸿章，我已经收到了你的邮件。谢谢！

FDwuyun, 2024-09-07:
一样的问题，我首先使仓库提供的vocab_larger.yml.gz，发现无法识别图1和图10的相似度。之后使用TUM数据集fr2_xyz和fr2_rpy通过gen_vocab_large.cpp重新生成vocab_lager.yml.gz，同样识别的效果不好，生成vocab_lager.yml.gz时的OpenCV版本我可以保证和loop_closure.cpp的一样。