gaoxiang12/slambook #271: make visualizeGeometry 报错

Mandylove1993, 2020-10-21:
安装并编译好了pangolin,但是make visualizeGeometry的时候报错：/usr/bin/cmake -H/home/zhuft/slambook/ch3/visualizeGeometry -B/home/zhuft/slambook/ch3/visualizeGeometry/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/zhuft/slambook/ch3/visualizeGeometry/build/CMakeFiles /home/zhuft/slambook/ch3/visualizeGeometry/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
make -f CMakeFiles/visualizeGeometry.dir/build.make CMakeFiles/visualizeGeometry.dir/depend
make[2]: Entering directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
cd /home/zhuft/slambook/ch3/visualizeGeometry/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/zhuft/slambook/ch3/visualizeGeometry /home/zhuft/slambook/ch3/visualizeGeometry /home/zhuft/slambook/ch3/visualizeGeometry/build /home/zhuft/slambook/ch3/visualizeGeometry/build /home/zhuft/slambook/ch3/visualizeGeometry/build/CMakeFiles/visualizeGeometry.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
make -f CMakeFiles/visualizeGeometry.dir/build.make CMakeFiles/visualizeGeometry.dir/build
make[2]: Entering directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
[ 50%] Building CXX object CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o
/usr/bin/c++   -I/usr/include/eigen3  -std=c++11   -o CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o -c /home/zhuft/slambook/ch3/visualizeGeometry/visualizeGeometry.cpp
[100%] Linking CXX executable visualizeGeometry
/usr/bin/cmake -E cmake_link_script CMakeFiles/visualizeGeometry.dir/link.txt --verbose=1
/usr/bin/c++  -std=c++11  -rdynamic CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o  -o visualizeGeometry 
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `main':
visualizeGeometry.cpp:(.text+0x5c2): undefined reference to `pangolin::CreateWindowAndBind(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int, pangolin::Params const&)'
visualizeGeometry.cpp:(.text+0x5f9): undefined reference to `glEnable'
visualizeGeometry.cpp:(.text+0x647): undefined reference to `pangolin::ModelViewLookAt(double, double, double, double, double, double, pangolin::AxisDirection)'
visualizeGeometry.cpp:(.text+0x6b2): undefined reference to `pangolin::ProjectionMatrix(int, int, double, double, double, double, double, double)'
visualizeGeometry.cpp:(.text+0x6d2): undefined reference to `pangolin::OpenGlRenderState::OpenGlRenderState(pangolin::OpenGlMatrix const&, pangolin::OpenGlMatrix const&)'
visualizeGeometry.cpp:(.text+0x6e1): undefined reference to `pangolin::CreateDisplay()'
visualizeGeometry.cpp:(.text+0x76e): undefined reference to `pangolin::View::SetBounds(pangolin::Attach, pangolin::Attach, pangolin::Attach, pangolin::Attach, double)'
visualizeGeometry.cpp:(.text+0x7a5): undefined reference to `pangolin::Handler3D::Handler3D(pangolin::OpenGlRenderState&, pangolin::AxisDirection, float, float)'
visualizeGeometry.cpp:(.text+0x7b0): undefined reference to `pangolin::View::SetHandler(pangolin::Handler*)'
visualizeGeometry.cpp:(.text+0xa9b): undefined reference to `pangolin::CreatePanel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
visualizeGeometry.cpp:(.text+0xb0b): undefined reference to `pangolin::View::SetBounds(pangolin::Attach, pangolin::Attach, pangolin::Attach, pangolin::Attach)'
visualizeGeometry.cpp:(.text+0xb2e): undefined reference to `pangolin::ShouldQuit()'
visualizeGeometry.cpp:(.text+0xb43): undefined reference to `glClear'
visualizeGeometry.cpp:(.text+0xb5c): undefined reference to `pangolin::View::Activate(pangolin::OpenGlRenderState const&) const'
visualizeGeometry.cpp:(.text+0xb6b): undefined reference to `pangolin::OpenGlRenderState::GetModelViewMatrix()'
visualizeGeometry.cpp:(.text+0xed9): undefined reference to `glColor3f'
visualizeGeometry.cpp:(.text+0xefb): undefined reference to `glLineWidth'
visualizeGeometry.cpp:(.text+0xf10): undefined reference to `glColor3f'
visualizeGeometry.cpp:(.text+0xf1a): undefined reference to `glBegin'
visualizeGeometry.cpp:(.text+0xf2b): undefined reference to `glVertex3f'
visualizeGeometry.cpp:(.text+0xf40): undefined reference to `glVertex3f'
visualizeGeometry.cpp:(.text+0xf55): undefined reference to `glColor3f'
visualizeGeometry.cpp:(.text+0xf66): undefined reference to `glVertex3f'
visualizeGeometry.cpp:(.text+0xf7b): undefined reference to `glVertex3f'
visualizeGeometry.cpp:(.text+0xf98): undefined reference to `glColor3f'
visualizeGeometry.cpp:(.text+0xfa9): undefined reference to `glVertex3f'
visualizeGeometry.cpp:(.text+0xfbe): undefined reference to `glVertex3f'
visualizeGeometry.cpp:(.text+0xfc3): undefined reference to `glEnd'
visualizeGeometry.cpp:(.text+0xfc8): undefined reference to `pangolin::FinishFrame()'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::_CheckGlDieOnError(char const*, int)':
visualizeGeometry.cpp:(.text._ZN8pangolin18_CheckGlDieOnErrorEPKci[_ZN8pangolin18_CheckGlDieOnErrorEPKci]+0x10): undefined reference to `glGetError'
visualizeGeometry.cpp:(.text._ZN8pangolin18_CheckGlDieOnErrorEPKci[_ZN8pangolin18_CheckGlDieOnErrorEPKci]+0x23): undefined reference to `pangolin::glErrorString(unsigned int)'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::GlTexture::~GlTexture()':
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTextureD2Ev[_ZN8pangolin9GlTextureD5Ev]+0x26): undefined reference to `pangolin::ShouldQuit()'
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTextureD2Ev[_ZN8pangolin9GlTextureD5Ev]+0x52): undefined reference to `glDeleteTextures'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::GlTexture::Bind() const':
visualizeGeometry.cpp:(.text._ZNK8pangolin9GlTexture4BindEv[_ZNK8pangolin9GlTexture4BindEv]+0x1b): undefined reference to `glBindTexture'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::GlTexture::Reinitialise(int, int, int, bool, int, unsigned int, unsigned int, void*)':
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x3b): undefined reference to `glDeleteTextures'
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x6e): undefined reference to `glGenTextures'
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0xba): undefined reference to `glTexImage2D'
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0xd8): undefined reference to `glTexParameteri'
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0xec): undefined reference to `glTexParameteri'
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x102): undefined reference to `glTexParameteri'
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x116): undefined reference to `glTexParameteri'
visualizeGeometry.cpp:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x12a): undefined reference to `glTexParameteri'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o:visualizeGeometry.cpp:(.text._ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv[_ZN8pangolin9GlTexture12ReinitialiseEiiibijjPv]+0x13e): more undefined references to `glTexParameteri' follow
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::GlBufferData::Free()':
visualizeGeometry.cpp:(.text._ZN8pangolin12GlBufferData4FreeEv[_ZN8pangolin12GlBufferData4FreeEv]+0x1a): undefined reference to `__glewDeleteBuffers'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::glDrawColouredCube(float, float)':
visualizeGeometry.cpp:(.text._ZN8pangolin18glDrawColouredCubeEff[_ZN8pangolin18glDrawColouredCubeEff]+0x490): undefined reference to `glVertexPointer'
visualizeGeometry.cpp:(.text._ZN8pangolin18glDrawColouredCubeEff[_ZN8pangolin18glDrawColouredCubeEff]+0x49a): undefined reference to `glEnableClientState'
[... cut ...]
CMakeFiles/visualizeGeometry.dir/build.make:94: recipe for target 'visualizeGeometry' failed
make[2]: *** [visualizeGeometry] Error 1
make[2]: Leaving directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/visualizeGeometry.dir/all' failed
make[1]: *** [CMakeFiles/visualizeGeometry.dir/all] Error 2
make[1]: Leaving directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

453875726, 2021-09-24:
> 安装并编译好了pangolin,但是make visualizeGeometry的时候报错：/usr/bin/cmake -H/home/zhuft/slambook/ch3/visualizeGeometry -B/home/zhuft/slambook/ch3/visualizeGeometry/build --check-build-system CMakeFiles/Makefile.cmake 0
> /usr/bin/cmake -E cmake_progress_start /home/zhuft/slambook/ch3/visualizeGeometry/build/CMakeFiles /home/zhuft/slambook/ch3/visualizeGeometry/build/CMakeFiles/progress.marks
> make -f CMakeFiles/Makefile2 all
> make[1]: Entering directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
> make -f CMakeFiles/visualizeGeometry.dir/build.make CMakeFiles/visualizeGeometry.dir/depend
> make[2]: Entering directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
> cd /home/zhuft/slambook/ch3/visualizeGeometry/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/zhuft/slambook/ch3/visualizeGeometry /home/zhuft/slambook/ch3/visualizeGeometry /home/zhuft/slambook/ch3/visualizeGeometry/build /home/zhuft/slambook/ch3/visualizeGeometry/build /home/zhuft/slambook/ch3/visualizeGeometry/build/CMakeFiles/visualizeGeometry.dir/DependInfo.cmake --color=
> make[2]: Leaving directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
> make -f CMakeFiles/visualizeGeometry.dir/build.make CMakeFiles/visualizeGeometry.dir/build
> make[2]: Entering directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
> [ 50%] Building CXX object CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o
> /usr/bin/c++ -I/usr/include/eigen3 -std=c++11 -o CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o -c /home/zhuft/slambook/ch3/visualizeGeometry/visualizeGeometry.cpp
> [100%] Linking CXX executable visualizeGeometry
> /usr/bin/cmake -E cmake_link_script CMakeFiles/visualizeGeometry.dir/link.txt --verbose=1
> /usr/bin/c++ -std=c++11 -rdynamic CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o -o visualizeGeometry
> CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `main': visualizeGeometry.cpp:(.text+0x5c2): undefined reference to `pangolin::CreateWindowAndBind(std::__cxx11::basic_string<char, std::char_traits, std::allocator >, int, int, pangolin::Params const&)'
> visualizeGeometry.cpp:(.text+0x5f9): undefined reference to `glEnable' visualizeGeometry.cpp:(.text+0x647): undefined reference to `pangolin::ModelViewLookAt(double, double, double, double, double, double, pangolin::AxisDirection)'
> visualizeGeometry.cpp:(.text+0x6b2): undefined reference to `pangolin::ProjectionMatrix(int, int, double, double, double, double, double, double)' visualizeGeometry.cpp:(.text+0x6d2): undefined reference to `pangolin::OpenGlRenderState::OpenGlRenderState(pangolin::OpenGlMatrix const&, pangolin::OpenGlMatrix const&)'
> visualizeGeometry.cpp:(.text+0x6e1): undefined reference to `pangolin::CreateDisplay()' visualizeGeometry.cpp:(.text+0x76e): undefined reference to `pangolin::View::SetBounds(pangolin::Attach, pangolin::Attach, pangolin::Attach, pangolin::Attach, double)'
> visualizeGeometry.cpp:(.text+0x7a5): undefined reference to `pangolin::Handler3D::Handler3D(pangolin::OpenGlRenderState&, pangolin::AxisDirection, float, float)' visualizeGeometry.cpp:(.text+0x7b0): undefined reference to `pangolin::View::SetHandler(pangolin::Handler*)'
> visualizeGeometry.cpp:(.text+0xa9b): undefined reference to `pangolin::CreatePanel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' visualizeGeometry.cpp:(.text+0xb0b): undefined reference to `pangolin::View::SetBounds(pangolin::Attach, pangolin::Attach, pangolin::Attach, pangolin::Attach)'
> visualizeGeometry.cpp:(.text+0xb2e): undefined reference to `pangolin::ShouldQuit()' visualizeGeometry.cpp:(.text+0xb43): undefined reference to `glClear'
> visualizeGeometry.cpp:(.text+0xb5c): undefined reference to `pangolin::View::Activate(pangolin::OpenGlRenderState const&) const' visualizeGeometry.cpp:(.text+0xb6b): undefined reference to `pangolin::OpenGlRenderState::GetModelViewMatrix()'
> visualizeGeometry.cpp:(.text+0xed9): undefined reference to `glColor3f' visualizeGeometry.cpp:(.text+0xefb): undefined reference to `glLineWidth'
> visualizeGeometry.cpp:(.text+0xf10): undefined reference to `glColor3f' visualizeGeometry.cpp:(.text+0xf1a): undefined reference to `glBegin'
> visualizeGeometry.cpp:(.text+0xf2b): undefined reference to `glVertex3f' visualizeGeometry.cpp:(.text+0xf40): undefined reference to `glVertex3f'
> visualizeGeometry.cpp:(.text+0xf55): undefined reference to `glColor3f' visualizeGeometry.cpp:(.text+0xf66): undefined reference to `glVertex3f'
> visualizeGeometry.cpp:(.text+0xf7b): undefined reference to `glVertex3f' visualizeGeometry.cpp:(.text+0xf98): undefined reference to `glColor3f'
> visualizeGeometry.cpp:(.text+0xfa9): undefined reference to `glVertex3f' visualizeGeometry.cpp:(.text+0xfbe): undefined reference to `glVertex3f'
> visualizeGeometry.cpp:(.text+0xfc3): undefined reference to `glEnd' visualizeGeometry.cpp:(.text+0xfc8): undefined reference to `pangolin::FinishFrame()'
> CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::_CheckGlDieOnError(char const*, int)': visualizeGeometry.cpp:(.text._ZN8pangolin18_CheckGlDieOnErrorEPKci[_ZN8pangolin18_CheckGlDieOnErrorEPKci]+0x10): undefined reference to `glGetError'
> visualizeGeometry.cpp:(.text._ZN8pangolin18_CheckGlDieOnErrorEPKci[_ZN8pangolin18_CheckGlDieOnErrorEPKci]+0x23): undefined reference to `pangolin::glErrorString(unsigned int)' CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pa [... cut ...]
> CMakeFiles/visualizeGeometry.dir/build.make:94: recipe for target 'visualizeGeometry' failed
> make[2]: *** [visualizeGeometry] Error 1
> make[2]: Leaving directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
> CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/visualizeGeometry.dir/all' failed
> make[1]: *** [CMakeFiles/visualizeGeometry.dir/all] Error 2
> make[1]: Leaving directory '/home/zhuft/slambook/ch3/visualizeGeometry/build'
> Makefile:83: recipe for target 'all' failed
> make: *** [all] Error 2

我也遇到了同样的问题，请问解决了吗？

453875726, 2021-09-24:
我遇到了同样的问题，解决了吗？

sq6er4, 2024-03-20:
I have the same problem,but it is gone after I make again.