gaoxiang12/slambook #274: 直接法推导泰勒展开问题

lanfeiying, 2020-12-06:
你好，我在读到直接法推导的时候对泰勒展开有点不理解。

原文如下：
[image]

我查了查barfoot state estimation for robotics，里面对非线性函数的李代数扰动描述如下：
[image]
[image]

在barfoot的推导中是没有使用链式法则（我的理解）。
我对直接法的理解与推导如下：
[image]

lanfeiying, 2020-12-06:
我后来想到后面还是可以做一步链式法则， 用复合函数的泰勒展开做，小量是一个复合函数，后面再接着把对\delta\xi的一阶项保留，但是有点不同的是书中雅克比矩阵的第一项dI2 / du是对扰动前三维点的导数，而不是扰动后的三维点的导数，不知道这样理解是否可行

closed 2021-03-22 by lanfeiying

GeoffreyHub, 2022-03-24:
说真的这里展开我也没看懂

PengKunPROO, 2023-04-10:
而且你有没有发现在最后你写的推导公式的第三行，把exp(\delta \xi)展开之后，然后乘开，式子的意义变了，第一项代表P在像素中的坐标没问题，但是第二项代表的意思就很迷惑，按道理是把P转换到相机坐标系之后再乘以一个小的扰动，相当于求了P在像素坐标系的坐标的周围的坐标，那这第一项和第二项一加肯定不符合对P进行微小扰动的求导结果啊，所以对应的意义我没搞清楚，您能帮忙解答吗