gaoxiang12/slambook #291: 12章loop_closure 出现Segmentation fault错误

caijun199702, 2022-04-18:
调试发现到db.add(descriptors[i])出错，但是不知道具体啥原因

Klngbob, 2022-10-09:
我也遇到了这个问题，请问你最后解决了吗？

Klngbob, 2022-10-09:
> 调试发现到db.add(descriptors[i])出错，但是不知道具体啥原因

我也遇到了这个问题，请问你最后解决了吗？

BTREE-C802, 2022-10-09:
您好，我叫旷鸿章，我已经收到了你的邮件。谢谢！

CJJuly, 2023-09-24:
出现这个问题加1

ksn13, 2024-01-03:
遇到了同样的问题，有什么解决方案吗？ 谢谢

ksn13, 2024-01-03:
我的解决方案：
      DBoW3::BowVector bowVec;
      DBoW3::FeatureVector featVec;
      // or with database 
      cout<<"comparing images with database :"<<descriptors.size() <<endl;
      DBoW3::Database db( vocab, false, 0);
✹     for ( int i=0; i<descriptors.size(); i++ )
      {
          vocab.transform(descriptors[i], bowVec);
          featVec.addFeature(i, 0);  // Adding features with default weight 0
  
          /* db.add(descriptors[i]); */
          db.add(bowVec, featVec);
      }
      
      cout<<"database info: "<<db<<endl;
✹     for ( int i=0; i<descriptors.size(); i++ )
      {
          DBoW3::QueryResults ret;
          vocab.transform(descriptors[i], bowVec);
          /* db.query( descriptors[i], ret, 4);      // max result=4 */
          db.query( bowVec, ret, 4);      // max result=4
          cout<<"searching for image "<<i<<" returns "<<ret<<endl<<endl;
      }
      cout<<"done."<<endl;