gaoxiang12/slambook #292: 对高翔博士“李代数与旋转矩阵的指数变换推导”质疑

JiangxinkeTao, 2022-04-20:
[image]

[image]

最后，诚盼您的回复，谢谢！

gaoxiang12, 2022-04-21:
这个事情并不是讨论从t_0到t之间的过程，而是讨论t_0处的导数。你谈的是如何在t_0到任意t之间的积分，但书里想谈的是在t_0处对R求导。
类比于一个标量函数f(t)，它在t_0处的导数为f'(t_0)，泰勒展开为：
f(t) = f(t_0) + f'(t_0) dt + O(dt^2)
这件事情在t_0的邻域内都是有效的。

书里的李代数 \phi 实际是角速度的定义方式，因为有 dR/dt = \phi R，所以在短时间内有 R=exp(\phi t)。当然长时间内这个瞬时速度是在变化的，自然需要积分。离散的积分就是你上面写的那种连乘的形式，参考IMU积分方程即可。
[image]

本节想介绍的只是\phi落在R(t_0)的切空间，以及切空间和流形之间的指数对数映射，积分这块倒没有太多介绍（因为没怎么提惯导）。

JiangxinkeTao, 2022-04-23:
[image]
[image]

高博士，以上是我的看法，请您批评指正。

gaoxiang12, 2022-04-24:
可以简化一下符号，一开始不用R(t)的写法，直接写dot(R)的公式。
[image]

JiangxinkeTao, 2022-04-24:
好的，OK，现在清晰了，您现在使用的方法本质上就是上述的一次代数精度机械求积公式得到的结果，非常谢谢你！感谢！

closed 2022-04-24 by JiangxinkeTao