gaoxiang12/slambook #295: g2o - csparse linker issue

mudit1729, 2023-01-31:
CMakeFiles/bundle_adjustment_g2o.dir/bundle_adjustment_g2o.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 9, 9, 0, 9, 9> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 9, 9, 0, 9, 9> > const&, double*, double*)':
bundle_adjustment_g2o.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x51): undefined reference to `g2o::csparse::CSparse::hasSymbolic() const'
bundle_adjustment_g2o.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0xa1): undefined reference to `g2o::csparse::CSparse::sparseView()'
bundle_adjustment_g2o.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x6ea): undefined reference to `g2o::csparse::CSparse::analyze()'
bundle_adjustment_g2o.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x72d): undefined reference to `g2o::csparse::CSparse::solve(double*, double*) const'
bundle_adjustment_g2o.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x76a): undefined reference to `g2o::csparse::CSparse::choleskyNz() const'
bundle_adjustment_g2o.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x993): undefined reference to `g2o::csparse::CSparse::writeSparse(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
bundle_adjustment_g2o.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0xcce): undefined reference to `g2o::csparse::CSparse::amd(g2o::csparse::CSparse::SparseView const&, Eigen::Matrix<int, -1, 1, 0, -1, 1>&)'
bundle_adjustment_g2o.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi9ELi9ELi0ELi9ELi9EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0xe91): undefined reference to `g2o::csparse::CSparse::analyze_p(int*)'
collect2: error: ld returned 1 exit status
CMakeFiles/bundle_adjustment_g2o.dir/build.make:97: recipe for target 'bundle_adjustment_g2o' failed
make[2]: *** [bundle_adjustment_g2o] Error 1
CMakeFiles/Makefile2:112: recipe for target 'CMakeFiles/bundle_adjustment_g2o.dir/all' failed
make[1]: *** [CMakeFiles/bundle_adjustment_g2o.dir/all] Error 2

While building ch9

BTREE-C802, 2023-01-31:
您好，我叫旷鸿章，我已经收到了你的邮件。谢谢！

MingHongDev, 2024-05-26:
I fixed it myself. The problem my cause by the link error between g2o and CSparse.
So, first install libsuitesparse-dev, then compile and make install g2o again.