gaoxiang12/slambook #304: 关于非线性优化的雅可比矩阵的疑惑？ch6和ch7

MaZhaoYong, 2024-12-21:
1. 在ch6中，给出了曲线拟合的手写高斯牛顿法，其中利用误差项e（x）对待优化变量x进行求导，得到的是雅可比矩阵J，也顺利代入增量方程 JJ^TΔx =-Je进行计算；”      H += inv_sigma * inv_sigma * J * J.transpose(); b += -inv_sigma * inv_sigma * error * J;“
2. 但是在ch7中，给出pnp的ba的手写高斯牛顿时，其中 利用误差项e（x）对待优化变量x进行求导，得到的却是雅可比矩阵的转置J^T，然后在程序代入增量方程时，先将J^T再装置一下。”      H += J.transpose() * J;  b += -J.transpose() * e;“
3. 我的疑惑时为什么两者同样是将利用误差项e（x）对待优化变量x进行求导，前者得到的是雅可比矩阵J，而后者得到的是雅可比矩阵的转置J^T
4.
[image]

BTREE-C802, 2024-12-21:
您好，我叫旷鸿章，我已经收到了你的邮件。谢谢！

MaZhaoYong, 2024-12-21:
[image]

MaZhaoYong, 2024-12-21:
[image]
[image]